miketeachman/micropython-rotary #6: needs pull-up

aa2mz, 2020-07-26:
Thank you , great work! My (cheap, 2 pin,24 count) encoders need a pull up to work. I did the pull-up before calling the routine and that stays set as long as the class'es Pin creation doesn't change it internally. (It doesn't currently.) Obviously must work without it in many cases so if you don't want to change it in the class, maybe just a note in the example?  Thank's again!

investigatronica, 2020-08-01:
like a charm

miketeachman, 2020-09-07:
I added a new constructor argument to handle encoder hardware that does not include pull-up resistors on CLK and DT
Commit:  https://github.com/miketeachman/micropython-rotary/commit/b5f2455b4841e8e7ad36706b6aaab28c8bba6104
Hopefully this change solves the issue

closed 2021-02-27 by miketeachman