miketeachman/micropython-rotary #15: Listener is being skipped sometimes

Karolis92, 2022-01-29:
Nice lib, it was really hard trying to handle all the states. But I encountered a problem, listener is not always called.
Using Raspberry Pico with program:
```
from rotary_irq_rp2 import RotaryIRQ

rotary = RotaryIRQ(
       pin_num_clk=6, 
       pin_num_dt=11, 
       min_val=0, 
       max_val=10, 
       reverse=False, 
       range_mode=RotaryIRQ.RANGE_UNBOUNDED,
       pull_up=True,
       half_step=False)

lastRotation = rotary.value()
def rotated():
    global lastRotation
    newValue = rotary.value();
    print(newValue)
    if lastRotation < newValue:
        print('clockwise ')
    else:
        print('counterclockwise ')
    lastRotation = newValue
    
rotary.add_listener(rotated)
```
If I edit ant print error instead of swallowing in rotary.py line 162, it prints `schedule queue full`
Maybe the problem lies in my rotary encoder taken from an old pc. When I tried writing my own code, it seemed to have a lot of small state changes that changes back before being read.

miketeachman, 2022-02-20:
Thank you for reporting this issue.  I have not yet had a chance to investigate, but it is on my to-do list!

miketeachman, 2022-10-08:
I finally fixed this problem !  commit [3522b829d91396a82ddd1fcd7c8e5c13ff5c793b ](https://github.com/miketeachman/micropython-rotary/commit/3522b829d91396a82ddd1fcd7c8e5c13ff5c793b)
Thank you for your patience.

closed 2022-10-08 by miketeachman