miketeachman/micropython-rotary #24: Use multiple rotary encoders with SN74HC165 as input expander

Guustaaf, 2023-03-01:
Hi Mike,

I have a project where I want to use five rotary encoders. There are of course GPIO expander ICs such as the MCP23017 but to use that kind of chip would bring in a lot of unnecessary complexity since all we need are inputs. The SN74HC165 is a commonly available 8 bit parallel-load shift register. So I imagine chaining two SN74HC165s together and then at short intervals reading the 16 bit value that contains the 15 bits representing the pins of the five rotary encoders. I could come up with some ideas for how to then link this to the micropython-rotary library, bit perhaps the smarter way to go is to let the author himself lead the way.

Guustaaf

miketeachman, 2023-03-03:
Unfortunately, a shift register will not work with this rotary encoder implementation.  This module requires pin interrupts built into each microcontroller to detect changes in the encoder value.  You might be able to find a different encoder library that uses polling techniques and then modify it to poll the shift register for changes in value.

Guustaaf, 2023-03-04:
OK, got it, thanks!