miketeachman/micropython-rotary #30: two encodes simultaneous on one esp32, one or other but not both work

eddo888, 2024-02-14:
Hi, I have a dual encoder (two encoders) connected to micropython on my esp32.
If I connect one or the other it works as expected.
however if I try to setup both at the same time, the first defined encoder works
the second one barly at all, ie it may work if you spin it real fast bot mostly it won't

is there some contention issue because they may be sharing interupts ?
please help.

samcwitte, 2025-11-24:
@eddo888 While not on an ESP32, I have a dual encoder with switch (a Bourns PEC11D-4120F-S0015) working with a Raspberry Pi Pico. I just represented it as two different encoders. If you'll see the datasheet for the encoder, there's A and A*. B and B*, etc. Those are for the two separate encoders. Check to see if yours has similar documentation. Code example below will change on your hardware setup. You may want to use pull up or pull down resistors on the encoder.

Code example:
```py
import time
from rotary_irq_rp2 import RotaryIRQ
from machine import Pin

# These are the GPIO pin numbers on my Pico. e.g. pinA1 is GPIO21, which is actually pin 27.
pinA1 = 21
pinA2 = 20
pinB1 = 14
pinB2 = 15

switch = Pin(13, Pin.IN) # Switch has external pull-down resistor
r1 = RotaryIRQ(pin_num_clk=pinA1, pin_num_dt=pinB1, half_step=True, pull_up=True)
r2 = RotaryIRQ(pin_num_clk=pinA2, pin_num_dt=pinB2, half_step=True, pull_up=True)

val_old1 = r1.value()
val_old2 = r2.value()

while True:
    val_new1 = r1.value()
    val_new2 = r2.value()

    if val_old1 != val_new1:
        val_old1 = val_new1
    if val_old2 != val_new2:
        val_old2 = val_new2
    # You can use switch.value() to see the current switch state

    print('val1 =', val_new1, " | val 2 = ", val_new2)
    time.sleep_ms(50)

eddo888, 2025-12-02:
Cool, thanks for getting back to me.
Merry Christmas !!

Yours Cheerfully

David Edson
+614 0868 0808
***@***.***